chrisdunnname/esphome-s3-box-3-lvgl #3: Device restarts when playing large audio files

zminde, 2025-07-27:
First, heartfelt thanks to the author of this branch for their work. This branch resolves the TTS voice playback issue with BigBobbas/ESP32-S3-Box3-Custom-ESPHome/box-3 in my environment. During several days of testing, I found that when the audio file generated by TTS is relatively large, it causes the device to restart. However, the same audio file plays normally under the official YAML configuration. I'm unsure if this issue can be addressed by modifying specific parameters in the YAML file. 
Note: The files causing the hardware restart are approximately 6MB, with a playback duration of about 3 minutes. I understand this is uncommon for typical use cases, but it can still occur in scenarios like natural disaster warnings or critical alerts issued by meteorological departments, as I am currently experiencing.

chrisdunnname, 2025-07-28:
I have a second thought for you. This could be ram/flash being low as decoding a large file would need this. My configuration includes two different animated UIs for voice assistant which means a lot of images which can use some RAM and the flash storage is currently over 90%. Depending on which UI you prefer you may be able to drop one of the options and related images/config to reduce size. 

For example to drop the Home UI you would comment out the following:
- lines 22-28 which start with anim2
- lines 1650-1687 from the set uimode script for Home mode
- lines 1721-1730 from the set uimode script for Home animation
- lines 2510-2544 for files for anim2
- listening page lines 5089-5093
- thinking page lines 5123-5127
- replying page lines 5173-5180
- error page lines 5239-5243
- line 6300 to remove option from dropdown

Similar for the hal animation just different lines in same sections of code. 
This will make the data much smaller and will reduce processing overhead so it might allow much larger files to be streamed. 

Just removing one of these modes should make a significant difference.

chrisdunnname, 2025-07-28:
I just released an updated config which trims a bit of weight. At the same time I added a "noanimation" config that removes all the voice assistant animations but keeps everything else. This should produce a heaps smaller firmware leaving more free flash storage and reducing image processing so this might stop your box rebooting.