chrisdunnname/esphome-s3-box-3-lvgl #5: Make Temperature Offset user configurable

chrisdunnname, 2025-08-28:
As the S3 temperature sensor is often slightly inaccurate requiring an offset it would be easier if the temp offset could be set via the UI or a control so that it could be easily retained and set differently per device without modifying the config.

BigBobbas, 2025-08-28:
Hi Chris filter and global below 
```yaml
sensor:
  - platform: aht10
    i2c_id: bus_b
    variant: AHT20
    temperature:
      name: "S3 Temperature"
      filters:
       offset: !lambda return (id(temp_offset));
      id: s3temp
      on_value:
        - lvgl.widget.show: temp_label
    humidity:
      name: "S3 Humidity"
      id: s3humidity
      on_value:
        - lvgl.widget.show: humidity_label
    update_interval: 60s
```
```yaml
globals:
  - id: temp_offset
    type: int
    restore_value: true
    initial_value: '0'
```

youkorr, 2025-08-28:
```
globals:
  - id: temp_offset
    type: float
    restore_value: true
    initial_value: '0'

number:
  - platform: template
    name: "Temp Offset"
    id: temp_offset_number
    min_value: -10
    max_value: 10
    step: 0.1
    restore_value: true
    optimistic: true
    on_value:
      then:
        - lambda: |-
            id(temp_offset) = x;   // update global
```

youkorr, 2025-08-28:
```
sensor:
  - platform: aht10
    i2c_id: bus_b
    variant: AHT20
    temperature:
      name: "S3 Temperature"
      id: s3temp
      filters:
        - lambda: |-
            return x + id(temp_offset);
      on_value:
        - lvgl.widget.show: temp_label
    humidity:
      name: "S3 Humidity"
      id: s3humidity
      on_value:
        - lvgl.widget.show: humidity_label
    update_interval: 60s
```

youkorr, 2025-08-28:
```
lvgl:
  pages:
    - id: main_page
      widgets:
        - spinbox:
            id: offset_spinbox
            min: -10
            max: 10
            step: 1
            value: !lambda 'return (int)id(temp_offset);'
            on_value:
              then:
                - lambda: |-
                    id(temp_offset) = (float)x;

        - label:
            id: offset_label
            text: "Offset °C"
            align: OUT_BOTTOM_MID
            x: 0
            y: 10
```

chrisdunnname, 2025-08-28:
Offset added as configuration option for temperature in 2025.08.28. 
This is not currently available as a UI option on the device as this still needs further work.

BigBobbas, 2025-08-28:
thats odd - looks like it must be a bug with the component.

BigBobbas, 2025-08-28:
tried with just setting an offset without a lambda and it works ok.

chrisdunnname, 2025-08-28:
Yeah.  I have always used an offset so this is odd. It seems that if there is a lambda in the component it has the problem. I wonder if it is an issue with how it compiles the sensors when it contains that. 
I had a crazy idea that I wonder if both sensors had a lambda it might work.  So if the humidity had a filter that was just x+ 0 then maybe they will compile the same way and work.

chrisdunnname, 2025-08-29:
OK. I took two steps forward and four back. 
It does not like the lamda referencing the global.

This works and both entities appear in HA.

```
sensor:
  - platform: aht10
    i2c_id: bus_b
    variant: AHT20
    temperature:
      name: "S3 Temperature"
      filters:
        - lambda: |-
            return x - 3.0;
      id: s3temp
      on_value:
        - lvgl.widget.show: temp_label
    humidity:
      name: "S3 Humidity"
      filters:
        - lambda: |-
            return x + 0.0;
      id: s3humidity
      on_value:
        - lvgl.widget.show: humidity_label
    update_interval: 60s
```

This causes humidity to disappear. 

```
sensor:
  - platform: aht10
    i2c_id: bus_b
    variant: AHT20
    temperature:
      name: "S3 Temperature"
      filters:
        - lambda: |-
            return x + id(temp_offset);
      id: s3temp
      on_value:
        - lvgl.widget.show: temp_label
    humidity:
      name: "S3 Humidity"
      filters:
        - lambda: |-
            return x + 0.0;
      id: s3humidity
      on_value:
        - lvgl.widget.show: humidity_label
    update_interval: 60s

globals:
  - id: temp_offset
    type: float
    restore_value: true
    initial_value: '0.0'
```

The state values are float to float so I can't see how there is any data format or conversion issue in the above. 

Potentially on startup it is getting an error if the number sensor sets the value as it may not pass a float. But that is for the temp not the humidity and temp is working fine. 

```
  - platform: template
    name: "Temp Offset"
    id: temp_offset_number
    entity_category: config
    initial_value: 0
    min_value: -10
    max_value: 10
    step: 0.1
    restore_value: true
    optimistic: true
    unit_of_measurement: 'C'
    on_value:
      - lambda: |-
            id(temp_offset) = x;
```

But this doesn't entirely make sense because BOTH sensors work on the device internally and values display on the screen. It is only that the humidity sensor no longer appears in HA.

BigBobbas, 2025-08-29:
i'll post in the ESPHome discord to see if anyone can shed any light. if no joy i'll raise an issue. It's certainly a head scratcher

chrisdunnname, 2025-08-30:
https://github.com/esphome/esphome/pull/10497

Looks like this is being fixed but if we move the globals above the sensor we can work around it now.

chrisdunnname, 2025-09-05:
This has been delivered. The final solution used the lvgl spinbox widget to set the offset, that was also implemented as a number component and used in the temperature filter.